egzumer/uv-k5-firmware-custom #384: Short transmissions prevent switching back to FM radio

KvallVall, 2024-01-06:
1. Set radio to memory mode
2. Enable FM mode
3. Make the radio receive a short transmission for the memory mode frequency
4. Occasionally the radio gets stuck in memory mode and never switches back to FM radio
5. If the radio receives more transmissions sometimes it unsticks itself and switches back to FM

egzumer, 2024-01-08:
I can't reproduce.

KvallVall, 2024-01-11:
Maybe it has something to do with the memory channel, it happens the most when I listen to inputs on a government repeater.

The memory for the repeater has RxCTCS, TxCTCS, and positive tx offset in the 440 MHz band. My radio is usually in Reverse mode for inputs only so my screen says CT L+R.

I can't remember if I set dual-watch on A/B to the same memory when I get this issue but sometimes I set it up like that.

commit 9472bbb5 (2024-01-12): Fix #384: FM radio getting stuck off on unrecognized incoming CTCSS tone
closed 2024-01-12 by egzumer