egzumer/uv-k5-firmware-custom #385: Scan List1 and List2 but not ALL channels

AubsUK, 2024-01-06:
Running Release v0.21

When a channel is stored, it can be added into List1 or List 2 or no list.

When scanning channels, you have an option of 1 (List1), 2 (List2) or * (ALL channels).  This concurs with the [instructions](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Radio-operation#memory-channels-scanning).

However, if I have a channel stored that I don't want to scan because it is constant static from my location (UHF repeater), but I want to have it in my channel list so I can hear it when I move locations, I can remove it from the List1, so when I scan List1 or List2, it is not scanned.  But if I scan *, it gets included (ALL channels).

**Question:** Is there a way to scan List1 (1) and List2 (2) but not ALL channels (*) ? - So it would have "1+2" or "12" in the top left.

egzumer, 2024-01-06:
No there is not

AubsUK, 2024-01-06:
Thanks for replying, appreciate it.

Is there any way to 'lock out' a channel ?

egzumer, 2024-01-06:
No, all is described in the wiki, there are no hidden features.

AubsUK, 2024-01-07:
Thanks, and sorry for asking.

Is this a limitation of the hardware itself, or could it be a feature through code? - Happy to look at the code to see if I can figure it out, unless you already know it's not possible.

JuantAldea, 2024-01-07:
It's purely software

egzumer, 2024-01-07:
`gEeprom.SCAN_LIST_DEFAULT` is what you should be looking for.

closed 2024-01-08 by egzumer

AubsUK, 2024-01-15:
Thanks for picking this up @egzumer

It took me a little while to figure the code out, but I've raised a pull request https://github.com/egzumer/uv-k5-firmware-custom/pull/420 for the suggested change:
https://github.com/egzumer/uv-k5-firmware-custom/pull/420