egzumer/uv-k5-firmware-custom #391: Nice to have : auto send dtmf every x seconds

Robby69400, 2024-01-09:
Hello,
could be a nice and hopefully easy feature, using the DTMF send fonction but instead of sending once could send every x s.
For example I leave my K5 somewhere sending this, I can check where it is received by moving with another talkie.

ErikS-web, 2024-01-10:
You known there is only 60 Kb for all the usefull functionality !

Suggest to make external hardware and connect it to the Kenwood-connector.

Much better idea....
Place your name on **poweron-screen** or in **your channellist**. So a finder might give it back

[image]

egzumer, 2024-01-12:
There is almost no practical use for the most part, so I'm not interested, but this is a nice little project to learn firmware modding...

Other than that you can try DTMF calling with respond function. It will call you back if you make a call with another radio.

closed 2024-01-12 by egzumer