egzumer/uv-k5-firmware-custom #394: S-Meter above S9

G8PTN, 2024-01-09:
A minor observation on the S-meter display.
When a receive signal is greater than S9, the “overS9dBm” value is only shown if it is >= 10 dBm over S9.
I sort of expected any level above S9 to be shown, but the behaviour is as per the code and perhaps as intended.
Many thanks for this excellent firmware upgrade for the UV-K5 series.
73, Dave

egzumer, 2024-01-10:
Initially the + values were shown in multiples of 10, like an analog smeter, there usually are 10dB ticks, later I decided that this limitatiin doesn't make sense. This is why it is like it is. I'm not sure this can be considered a bug, it just works how it works. Maybe switching to + values after 6dB would be better, but does it really matter?

G8PTN, 2024-01-10:
Hi,
Thanks for the response. All understood and accepted.
I will close the item.
73, Dave

closed 2024-01-10 by G8PTN