egzumer/uv-k5-firmware-custom #395: sql calibration implementation as in original and ijv firmware

pawol, 2024-01-09:
please take into account an implementation of existing sql calibrations read from eeprom like in the original or ijv firmware. each band has sepatared sql calibration jept in the eeprom. for instance sql levels are shifted 14dBm between uhf and vhf nbands

egzumer, 2024-01-09:
use chirp

closed 2024-01-09 by egzumer

pawol, 2024-01-10:
it is not possible to recalibrate sql eeprom values by Chirp. Just battery and tx power

pawol, 2024-01-10:
[image]

egzumer, 2024-01-10:
You dont need chirp developer options
https://github.com/egzumer/uvk5-chirp-driver?tab=readme-ov-file#calibration-settings

egzumer, 2024-01-10:
And I don't know where did you get that driver from, that is not my driver