egzumer/uv-k5-firmware-custom #396: tx power calibration implementation suggested

pawol, 2024-01-09:
please take into account an implementation of tx power calibrations read from eeprom like in the original or ijv firmware. each band has sepatared tx calibration kept in the eeprom. for instance tx level can be different between uhf and vhf bands . The issue can be easily investigated when ijv firmware is used, Low, med, high levels are changed and eeprom is dumped

egzumer, 2024-01-09:
use chirp to change calibration values

closed 2024-01-09 by egzumer

pawol, 2024-01-09:
you do not understood. Every L M H setting has own value in the eeprom. You can set eg. 0,5W for Low power on 2m band. You have to try ivj. Chirp has nothing common with this issue

egzumer, 2024-01-09:
You don't understand, use calibration section in the chirp

pawol, 2024-01-09:
Sadly I see here same regres like with 1of11

egzumer, 2024-01-09:
What regres? Did check out the chirp driver or not? I guess not.

egzumer, 2024-01-10:
Battery calibration is a different thing, it is one option and not gezilion different options like in the sql, or txp. Battery is easy. For TXP you need power meter and you need to know what youre doing, if someone whants to mess with that you have that option. I'm not adding that to the menu