egzumer/uv-k5-firmware-custom #399: Show Modulation Settings

ludwich66, 2024-01-10:
Please display  FM + (AM, USB) modulation also on Screen.

ErikS-web, 2024-01-10:
? FM is the default of this FM-radio !

For AM and USB it's only demodulation (=RX) and **No TX !**

ludwich66, 2024-01-10:
I don't fully understand the answer?
The radio displays AM (amplitude modulation) in aviation radio or USB (sideband modulation). Now I would also like to see “FM” (frequency modulation) in the display. What do you mean just RX or TX?

ErikS-web, 2024-01-10:
Nothing on display is the default "FM"

The radio does not modulate AM or USB (that means TX) and only receives in those added modes. (that is called **DE**modulation)

magellan-13016, 2024-01-16:
This would be effectively nice to display FM when the demodulator is in FM mode.

egzumer, 2024-01-16:
This is an FM transceiver, FM is default. AM and USB are stub modes that are not TX capable. They are not equal in functionality to FM, so it makes sense to show AM/USB, it doesn't really make sense to show FM. This is how it was implemented from the start by quansheng. There is less clutter on the screen, I like the way it is, and it also means I don't have to do anything if I leave it as is, which is a plus.