egzumer/uv-k5-firmware-custom #400: Enhancement/question: Lock transmission to PMR446 frequencies only

4rianton, 2024-01-10:
I am very new to the hobby, so please excuse any mistakes I make with the description. 

I got a UVK5 and being unlicensed, I am afraid of accidentally Tx on a frequency that I am not allowed to Tx on. Therefore, I tried to look for an option in Hidden Menu 61 that only allows PMR446 Tx but wasn't able to find it. 

Is there an easy way to do so? Or is the only way to do it is to add an option manually in code and then build the firmware that way? 

Thanks!

dadanielat, 2024-01-10:
You will have to define all PMR channel frequencies using CHIRP (be sure to use NFM mode and low power) and disable 'frequency mode allowed' in CHIRP advanced settings.

[image]

ErikS-web, 2024-01-10:
That is the way to do it :+1:

4rianton, 2024-01-10:
> You will have to define all PMR channel frequencies using CHIRP (be sure to use NFM mode and low power) and disable 'frequency mode allowed' in CHIRP advanced settings. You can also add 'listen only' frequencies with duplex set to off, then TX for this channel will be disabled.
> 
> 
> 
> Download from: https://github.com/egzumer/uvk5-chirp-driver/releases
> 
> 
> 
> [image]
> 
> 

Thank you for your response. I was able to download and install CHIRP and egzumer module driver to it and was able to create memory entries for PMR frequencies. 

Is there an easy way to be able to still Rx on all the other frequencies that this radio can receive? 

You mentioned that I can add frequencies in the memory list with duplex off but that will still be limited to whatever exact frequencies I add which will be pretty limited. Am I correct in my assumption?

dadanielat, 2024-01-10:
AFAIK it is currently not possible to disable TX in frequency mode. You only have the choice of disabling frequency mode entirely. This should be raised as feature request. I know this setting is available in opengd77 firmware, but this is made for other radios.

ErikS-web, 2024-01-10:
Menu 61. `F lock` and select DISABLE ALL - disables TX on all frequencies

4rianton, 2024-01-10:
> [Menu 61. `F lock` ](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Menu#hidden-menu)and select DISABLE ALL - disables TX on all frequencies

I checked that out. Unfortunately, I would like to still be able to TX on PMR frequencies but RX on all the other ones. Does it look like the only way to go about it is to update the code to add PMR entry to Menu 61 and roll my own build?

gartnl, 2024-01-11:
@4rianton I do not completely understand your question. If you are unlicensed you are not allowed to transmit anywhere. The device is illegal on PMR so nobody ( not even  HAM's ) can use it on PMR.
In fact the only legal use is in the US on HAM frequencies. And some countries where they don't bother at all.

dadanielat, 2024-01-11:
gartnl is correct, only devices with pre-programmed immutable channels and without detachable antenna are legally allowed for PMR use. However, if programmed correctly, hopefully nobody should notice a difference :)

tonurics, 2024-01-11:
> @4rianton I do not completely understand your question. If you are unlicensed you are not allowed to transmit anywhere. The device is illegal on PMR so nobody ( not even HAM's ) can use it on PMR. [...]

As PMR446 is a licence-exempt set of bands within the European Union. You're U.S. centric comments are not entirely correct; i.e. not being band licensed.

However, fact that the antenna is removable invalidates this radio on PMR446. And another problem for PMR446 is that the transmit power limit of 0.5w; which is well below the stock low output level. Ref. [ECC PMR446 Requirements](https://docdb.cept.org/download/1491)

gartnl, 2024-01-11:
@tonurics Funny, since I am Dutch. 

PMR446 needs no license but is off-limits to this set no matter what. I never and nowhere stated you need a license for PMR. I do however meet HAM's who believe that just because they have a license they can use PMR with HT's like this. They can't. Just like I'm not allowed to use this/any HT to talk to LPD users. Or use my callsign on PMR. 

The HT is illegal for TX in the EU anyway. With harmonics/spurious at around -45 dBc on HAM bands at best it could be legal on HAM bands in the US. All other frequencies are off-limit by default because they don't even meet the FCC requirements. 

BTW: most (no-name) China PMR stuff is illegal in the EU as well. Non-ham equipement has to have a harmonics/spurious output of no more than -36 dBm. See professional PMR sets specs .

4rianton, 2024-01-11:
Thank you for all this information. Very eye opening.  

It really seems like I should keep the TX off on this radio and really look into getting my Amateur radio license if I want to have any sort of fun with this hobby.

closed 2024-01-11 by 4rianton