egzumer/uv-k5-firmware-custom #401: display questions

VintageEngineer, 2024-01-10:
[image]
What are the circled indicators on the display?

```
R - is this redundant with the bold RX or does it mean something else?
><  - ?
I  - ?
```
addendum:
Here is the stock screen description:
[image]
"R" sometimes means "Reverse" but my radio wasn't set in Reverse mode. The "><" and "I"  are mysterious to me as well.

Might be nice to have a little guide for all the display icons for us newbies?

dadanielat, 2024-01-10:
R seems to be redundant, but RX could be displayed up on A or down on B channel. '1 ><' or '2 ><' or '* ><' is displayed if scan mode/scan list is active.

VintageEngineer, 2024-01-10:
> R seems to be redundant, but RX could be displayed up on A or down on B channel. '1 ><' or '2 ><' or '* ><' is displayed if scan mode/scan list is active.

R is confusing as it often means "Reverse". I don't understand your comments on the scan list... as the "><" symbol is at the top, not associated with one channel or the other? And the "I"?

VintageEngineer, 2024-01-10:
Ok, I think the "R" and "T" is a General and redundant "Transmit" and "Receive" indicator.
I think these could be deleted and the space used for other stuff, given the giant "Rx" and "Tx" that also exists and is dominant.
The "><" and "I" are still mysterious to me...

ErikS-web, 2024-01-10:
Don't use the QS manual.

We've made an [Online Manual in the Wiki](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Radio-operation#memory-channels-scanning)

VintageEngineer, 2024-01-10:
> Don't use the QS manual.
> 
> We've made an [Online Manual in the Wiki](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Radio-operation#memory-channels-scanning)
Ok, so the "I" is for scan group "1"? 
Thank you!
Suggestion... Maybe consistently use Roman or Arabic numbers?
Any ideas on the "><"?

ErikS-web, 2024-01-10:
👍 

The >< is always 4 seconds and what is after that on display. Probably DW. So DW will continue after the waittime, where it stays only on this VFO.

egzumer, 2024-01-10:
The >< means dual watch is disabled