egzumer/uv-k5-firmware-custom #414: Display Changes Requested

VintageEngineer, 2024-01-13:
Request changes to display:

1) **Delete extra (redundant) “R” and “T” in the upper left corner of the display that indicate receive and transmit.**

2) **Make the scan group indicators consistent.  Call them “I” and “II” on the screen in both the top left while scanning, to the right of the channel label, and in the manual.**

3) “><” indicates the response timer is running for “dual rx respond”. But it erroneously also shows when in “main tx dual rx”.
Current implementation:
**dual rx respond**: shows "DWR" and then shows "><" for 4 seconds after a transmit or receive
**main only**: shows nothing
**main tx dual rx**: shows "DW" and then "><" for 4 seconds after a transmit or receive
**crossband**: shows "XB" all the time

**I suggest deleting the “><” indicator and replace with two “>” indicators, one by each channel, just as done in the stock firmware. This indicates not only the respond 4 seconds is active, but also what channel will be responded to.**

ErikS-web, 2024-01-29:
Why change. Isn't it QS-default ?

KDMcMullan, 2024-02-08:
> Why change. Isn't it QS-default ?

The AM bug is QS default. Change for the better. Removing this redundancy might even give us a few more bytes free code space.