egzumer/uv-k5-firmware-custom #425: No audio when using the NOAA feature

travisbeale, 2024-01-19:
The feature that allows NOAA frequencies under their own set of channels is very useful here as these stations transmit continuously.  In order to enable it, I set ENABLE_NOAA = 1 and ENABLE_SPECTRUM = 0 in the Makefile.  F+5 enters NOAA mode and the green light on the radio lights up indicating a signal.  However, I do not hear any audio unless I enable Monitor mode.

baumertjohn, 2024-01-19:
I believe this is the way the "stock" firmware works.  The light lets you know it's monitoring and audio will come on when there is an alert sent out.

travisbeale, 2024-01-19:
I did play with the stock firmware a bit and don't recall it working that way but that explanation makes some sense.  In that case, this is more of a feature request to be able to disable the "alert" detection and just listen to the channel.  Monitor mode kind of does this, but then you can't scan for whatever your local NOAA channels are.

baumertjohn, 2024-01-19:
Yeah, seemed like an odd feature.  I ended up just adding the channels in memory as "no-transmit" to use them.

egzumer, 2024-01-20:
There is no NOAA in Europe where I live, so I don't have any interest/use for NOAA, neither have any way to test this. I didn't investigate that but from what I remember the squelch is deactivated when it receives particular tone (1kHz?) which is probably an alert tone I would assume.

travisbeale, 2024-01-22:
I can test.  Could you point me to the place in the code that listens for this tone?

egzumer, 2024-01-22:
https://github.com/egzumer/uv-k5-firmware-custom/blob/c74c0d20fc0e2efe54fbf4ea108139c2b0c36502/driver/bk4819.c#L474
https://github.com/egzumer/uv-k5-firmware-custom/blob/c74c0d20fc0e2efe54fbf4ea108139c2b0c36502/radio.c#L671