egzumer/uv-k5-firmware-custom #431: Scrambler frequency below 3100 not usable

mentanah, 2024-01-21:
Programmed three radios with v0.21.
Set channel with scrambler on to 2900Hz, for each radio at  446.08125 MHz.
Now you can her a loud hissing noise while talking, approximate comparable like wrong selected scrambler frequency. Adjusting it above 3200 Hz on all devices results in clear scrambler decoding.

ea7hjb, 2024-01-28:
I noted this problem since the first days of Quansheng, even with the original firmware. I tested it vs many brands (Motorola, Vertex, etc) and while I haven't been unable to reproduce exactly when or how it occurs, it happens a lot.

I think it is related to the CTCSS/DCS, as you can hear the subtone after the inverter processing as a continuos "biiiip" if analog, or "bip bip bip" if DCS. If you can, repeat the test with the transmitter using no subtone at all.

mentanah, 2024-02-01:
> I noted this problem since the first days of Quansheng, even with the original firmware. I tested it vs many brands (Motorola, Vertex, etc) and while I haven't been unable to reproduce exactly when [...]

Shure, that's it! Thank you for this good explanation. Since the DTMF must not be scrambled, it is not "encoded" but "decoded" as well.

mentanah, 2024-02-01:
Closed as good explanation is delivered.

closed 2024-02-01 by mentanah