egzumer/uv-k5-firmware-custom #432: Version number

HotelMikeSierra, 2024-01-21:
Sorry if this was answered elsewhere, new at this compiling thing.
Compiled the firmware with Docker, followed the instructions, but the firmware displayed is "EGZUMER 46e2d1" not v0.21.
Don't know if I’m doing something wrong.

Thank you.

[image]
[image]

Tyrbiter, 2024-01-21:
It's just a git hash value, it shows the latest commit used to build the release known as v0.21

egzumer, 2024-01-21:
You will get the actual version only if you use my official release build, it is automatically set with github action. If you want to set it on your own you have to play with a VERSION_STRING in the makefile

closed 2024-01-21 by egzumer