egzumer/uv-k5-firmware-custom #434: Squelch Tail Elimination (STE).

slavikby, 2024-01-21:
Hello. I noticed that the station transmits a Tail tone while transmitting. (An indirect sign of the transmission of this tone is that when the PTT is released, the station continues to transmit for several milliseconds). The STE menu item does not disable it. Is it possible to make it switchable? Thank you !!!

OE1MWW, 2024-01-22:
[of](https://forums.radioreference.com/threads/squelch-tail-elimination-ste.341657/) 2016, but nothing has changed

slavikby, 2024-01-22:
Thank you. That's exactly what it is. I'm wondering if it's possible to turn off this signal during transmission. When changing the STE menu item on the current firmware, the signal is retained.

commit a22760c0 (2024-01-29): Fix #434: Don't send tail tone when STE is off
closed 2024-01-29 by egzumer