egzumer/uv-k5-firmware-custom #437: Around 140.000 tx disabled after set band tx on

wkdake, 2024-01-23:
Hi,

I set the band of 137-174MHz TX ON by all means that instructed on your wiki site (including choosing 'unlock all' 10 times), but it still could not tranmit in that freq range. It shows "tx disable". But it can transmit on 400-470MHz. I purchased this Quansheng K6 from China two days ago.

Thanks a lot!
Regard!

wkdake, 2024-01-23:
i flashed back to the orginal Quansheng firmware, problem still. then i found that i set TX CTCS -8mhz, so i reset all, then in original firmware i finnally can tx on 137mhz. then i flashed agian egzumer firmware, and reset all again, chose unlock all 10 times again----Now my mind is in peace. thanks!

wkdake, 2024-01-23:
So i guess, 130mHz tx is the hardware low limitation there, no matter how hacked on display, it won't work under that properly. am i right?

egzumer, 2024-01-23:
You shouldn't transmit on 130MHz under any circumstances!!! This is airband (108-137)!

closed 2024-01-23 by egzumer

wkdake, 2024-01-23:
> You shouldn't transmit on 130MHz under any circumstances!!! This is airband (108-137)!

Thanks for remindering, well noted. i tried transmission in my safe box door-semi-opened, hopefully did't bother anything above

egzumer, 2024-01-23:
This radio is not AM capable and TX in AM mode is locked so nobody tries either way.