egzumer/uv-k5-firmware-custom #442: [ENHANCEMENT] DLive show over RSSI bar

kamilsss655, 2024-01-26:
Currently the radio allows to send DTMF codes by pressing number buttons during pressed PTT, and the receiving party can see these codes using the DLive functionality.

This can be useful to send some information digitally.

However now when DLive is enabled the received codes are not displayed, because the RSSI bar covers it and are only shown when RX ends.

It would be cool that if DTMF codes are received they are displayed immediatelly and the RSSI bar is hidden for the duration.

WHYT @egzumer ?

egzumer, 2024-01-26:
Radio receives a lot of garbage DTMF codes. It will hide the rssi. For me personally live DTMF is useless, it is only there because it was added by 1o11 and it didn't bother me.

kamilsss655, 2024-01-26:
Yea that's a fair point.

I was thinking the same about the usefulness of it, but the ability to send digital data now makes it appealing to me.

Closing this one, as indeed that could hide RSSI bar when unintended.
I might just make the delay longer so the codes are displayed longer.

closed 2024-01-26 by kamilsss655