egzumer/uv-k5-firmware-custom #447: Foibles after flashing v0.22

G8VJN, 2024-01-30:
1) I see no facility within the modified CHIRP for set the action for programmable LONG PRESS Menu button. This requires re-setting after every memory upload to radio.

2) After flashing v0.22, when in memory mode on either or both VFO's there is no mode of operation visible.

3) The facility to UNLOCKED ALL for TX, requires re-setting after every memory upload to the radio. As it is such a pain to do, this is, for me, a major issue.

Other than these foibles, great work and most impressive.

Many thanks... Len Gurney, G4LBJ

ErikS-web, 2024-01-30:
Seems to me these are 3 issues...

1 woud be an issue on Chirp. And should be adressed [here](https://github.com/egzumer/uvk5-chirp-driver/issues)
2 also, because FM is default. Did you test for that ?
3 is intented to work so. Can only be for test-purposes, so OFF after each power-on would even be better.

f4axk, 2024-01-30:
chirp_egzumer0.81
programmable LONG PRESS Menu button

[image]

G8VJN, 2024-01-30:
Thank you for your input. Sarcasm not required, nor appreciated.

Thank you.
Len Gurney

egzumer, 2024-02-01:
Read the warning note:
https://github.com/egzumer/uvk5-chirp-driver?tab=readme-ov-file#how-to-use

closed 2024-02-01 by egzumer