egzumer/uv-k5-firmware-custom #449: Remote Speak/Mic issue

G8VJN, 2024-01-30:
It would be really good for the standard "Baofeng" remote speak/mic would work with this firmware as it did on the standard factory firmware. I am aware there is mention of this in the paperwork, BUT!!

For ME, this is a major draw back as it will prevent me using this radio on portable events and when out and about.

Is there any chance of a fix please.. pretty please.. with knobs on !!!!

Thank you.. Len

ErikS-web, 2024-01-30:
I do not understand... Is this a **firmware/software issue** ?

This Github is for EGZUMER firmware for the radio

If not than please close the issue.

G8VJN, 2024-01-30:
It is a firmware issue, for if I chsnge firmware,  all is good.

Thank you for your input.

Len Gurney

G8VJN, 2024-01-30:
> what do you mean by 'remote speak/mic/ ? what does not work for you
> Have a look here
> https://github.com/egzumer/uv-k5-firmware-custom/discussions/267
> <http://url>

A remote speaker/mic is just that, a combined soeaker and microphone with
ptt attached to a curly cable that plugs into the side of the radio and
allows you to hsve the radio clipped to your belt, and the speaker/mic on
your lapel.

Ok?

Cheers

Len Gurney

G8VJN, 2024-01-31:
I have edited the issue description, to clarify this is not a hardware issue but a firmware one.

Thank you.

ErikS-web, 2024-01-31:
How can this be... 

And  **_explain what is working and what not._**

**You have made nothing clear at this moment !**

An external microphone is nothing more than x functionalities
- PTT switch
- Microphone
- Speaker
- optional PTT-LED
- optional earphone connector

What type of microphone do you using? 
Make a photo and reply.

This is all hobby for us, so if you want help. Would be nice to read full information

OE1MWW, 2024-01-31:
@G8VJN, you complain about snotty email, but you did not explain your working conditions

- it's a 'speaker/mic' and not a 'speak/mic' and this device is not 'remote', because it is connected to the device.
- you never told us, do you use a UV-K5 or a UV-K5(8), sometimes called UV-K6
- on the UV-K5(8), there is a known hardware bug. See atached required capacitor

so - what is your problem now ? what and how does it not work in your setup ?

[image]

OE1MWW, 2024-02-01:
btw, my UV-K5(8) with the shown speaker/mic as in 
https://github.com/egzumer/uv-k5-firmware-custom/discussions/267#discussioncomment-7855807
works perfect with the v0.22 release - so what's up with yours ?

G8VJN, 2024-02-01:
> now you have posted 5 entries plus a picture - but still not explained what 'does not work' ?
> Dont complain about 'snotty emails', a title like 'remote speak/mic' is not a common term.
>
> There is a known hardware issue (bug), as long as squelch is open, (signal is present) you
> can not PTT via the not so 'remote connected' speaker/mic - is this your problem ?
> Had been discussed in another thread in detail.

The issue is described.. please read again..

G8VJN, 2024-02-01:
> and this modified speaker/mic works on an Kenwood TH-D74E without any glitches.
> Same with any other firmware on the UV-K5(8) - no problems seen.
>
> Your 'This then could render the device Un-serviceable on other radios' is not the case.

Please provide evidence of this, for it did not on others of my hand held radio’s

G8VJN, 2024-02-01:
Yes, as described in the edited issue..as for a previous discussion, I neither looked for nor saw such a discussion. I find GITHUB confusing at my age.

OE1MWW, 2024-02-01:
"Please provide evidence of this,"  no, sorry, I give up in this thread
Has been discussed in detail here https://github.com/egzumer/uv-k5-firmware-custom/discussions/267

egzumer, 2024-02-01:
This IS a hardware issue. You've been pointed to the cause and the solution.

closed 2024-02-01 by egzumer

G8VJN, 2024-02-01:
Rubbish.... One should not need to modify hardware to work with firmware... I remson unsure what planet you are on... Seeing as YOU closed the issue, we will never get the correct remidial action instigated. Thanks a bunch.

egzumer, 2024-02-01:
> One should not need to modify hardware to work with firmware...

This won't work on any firmware. Firmware is unable to distinguish between internal or external PTT.

G8VJN, 2024-02-01:
Yes it does.. It worked perfectly on the stock firmware on this very K5(8)....as it works on two other radios..... You have no Idea what you are talking about sir.

egzumer, 2024-02-01:
Show a video of how it works.

egzumer, 2024-02-01:
And to be clear, I wan't you to show how it works on the same radio on which it doesn't work with my firmware. Flash a stock firmware back on this radio and do the test.

G8VJN, 2024-02-01:
I wish I could flash the original firmware back on this radio.....It seems to defy my attempts to do just that. Anyway sir, my treatment on this platform for trying to raise an issue has been deplorable, and consequently I have washed my hands of this and am moving on.I applaud your inginuity and intellect, and congratulate you on your achievement, but sadly your followers are a inconsiderate and bullying bunch, yes, bullying me in writing, at least that is how it felt, nay, feels to me, and being 72 years old, I know bullying when it happens. I hope they are proud of themselves.Kindly do not bother yourself further sir, for your bully boy followers have succeeded in driving me away and I shall try some other person's genius instead, and report back on the user group forum of my treatment on this platform. I wish you well, with peace and happiness. Just try to keep the dogs under control please.Kind regards, Len Gurney ISM

G8VJN, 2024-02-01:
You are mistaken, for when on a busy of open channel, and the external pgg does not operate, the built in pgg still does, so the firmware IS able to distinguish between the two.Len

egzumer, 2024-02-01:
Well, one can't always be right. I'll accept my defeat this time.

G8VJN, 2024-02-01:
Forget it sir, life is too short for regrets.

Have a good life.

L W Gurney ISM

bsdinis1, 2024-02-21:
> You have no Idea what you are talking about sir.

You have no idea what you are talking about, sir.
There is a DC voltage present on the external speaker output when the amplifier is turned on, and the radio only turns on the amplifier when it needs to output sound.
The GPIO for the PTT functionality must be pulled low to register as a key press, and as you can imagine having ~4V DC present on that GPIO pin might be a problem...

> 72 years old

Yes, and it clearly shows..
Multiple people have tried to explain that this is a hardware issue but you insist on blaming the firmware which makes absolutely no sense.
What happened is; when you tried to use the PTT on the external speaker/mic using the stock firmware the radio did not have the amplifier turned on, in other words it wasn't outputting sound which caused no problems, but when you tried to use the external PTT with the custom firmware the DC voltage present on the amplifier's output did not allow the GPIO to be pulled low.

G8VJN, 2024-02-21:
Yes I am 72 years old, and yes it does show, in more way than one. Nevertheless  at least I have managed to learn and preserve the basics of respectful and reasoned communication. 

Your insolent and antagonistic trolling is not appreciated. However the technical content is; for you have qualified better than I ever could, that this firmware differs from the original (or “stock”),  in the way that it manages the external speaker/mic port. 

You explain that the “stock” firmware does NOT have the amplifier (?) turned on, whereas the “custom” (this) firmware does, so thank you,  Mr/Mrs/Ms mysterious hidden behind keyboard warrior, you have just PROVED my case, that the difference in external speaker/mic management IS the reason for the issue.

I had actually let this issue drop after an admin told me he had closed the “case”. I was comfortable with that, as I do not appreciate silly little, partially educated people  being boorish, invasive, discourteous and disrespectful. I do however, thank you for proving my case which I had dismissed as an insignificance.

Kind regards,

G8VJN
L Gurney, ISM

egzumer, 2024-02-21:
> You explain that the “stock” firmware does NOT have the amplifier (?) turned on, whereas the “custom” (this) firmware does

He didn't proove anything because his description is wrong in that part. You see bulies evereywhere around you, there is something wrong with your eyesight. You're the one who is impossible to reason with, you know better no matter what.

G8VJN, 2024-02-21:
Perhaps that comes with age, experience and, oh yes, knowledge and in this
case particularly, no ego to feed, what ho.

egzumer, 2024-02-21:
Oh, so you came just to mentor us, thanks a lot.

what a joke 🤣🤣🤣

bsdinis1, 2024-02-21:
> You explain that the “stock” firmware does NOT have the amplifier (?) turned on, whereas the “custom” (this) firmware does, so thank you, Mr/Mrs/Ms mysterious hidden behind keyboard warrior, you [...]

Why did I even bother try to explain anything to you...?
I'll accept my defeat, there is no arguing with you... this feels like trying to explain to a monkey how a car works.

bsdinis1, 2024-02-21:
> He didn't proove anything because his description is wrong in that part. You see bulies evereywhere around you, there is something wrong with your eyesight. You're the one who is impossible to [...]

you mean HIS description?
I definitely did not say that "the stock firmware turns on the amplifier" and that the "custom firmware does not".

egzumer, 2024-02-21:
> you mean HIS description?
> I definitely did not say that "the stock firmware turns on the amplifier" and that the "custom firmware does not".

I meant yours. You didn't say that, and actually you can be right, but a solder blob is more likely cause.