egzumer/uv-k5-firmware-custom #455: Optional increase sensitivity in the Satcom band.

slavikby, 2024-02-02:
Hello !
Thank you for the development and improvement of the firmware!
Is it possible to optionally increase sensitivity in the Satcom band?

slavikby, 2024-02-03:
I understand the question, but I will say that on one of the current alternative firmware for K5, I received Satcom, standing at the window, the radio without modifications. The description of the tested firmware mentioned increased sensitivity. Those. I propose to reveal the AVAILABLE POTENTIAL of the sensitivity of the receiver on this band.

ErikS-web, 2024-02-03:
And HOW do we do that? _Hokusspokus,_

If you can figure out the HOW, it could possibly be done in the firmware.
Without that data I think....

slavikby, 2024-02-03:
I believe this is due to raising the gain level of the chip's internal circuits at certain frequencies.
This is implemented in the IJV firmware. I installed it for a test and immediately noticed the better sensitivity on Satcom.

ErikS-web, 2024-02-03:
Well _ask them_ how they did it.

That project is NO open source,  so we don't known.

slavikby, 2024-02-03:
"Activates the filter change circuit for frequencies after 240 MHz and sets the radio chip to a +9dB increase in reception, this once activated remains in memory the next time it is switched on. Similar to a preamplifier, to be used in exceptional cases of low signal because it can cause distortion in listening, switch it off when not needed."

mentanah, 2024-02-03:
You can try the following: go to spectrum, press PTT to open manual gain settings, open squelch, set right frequency, play with LNA and PGA gains manually and crank up sensitivity. Please report here...

slavikby, 2024-02-07:
**mentanah**, the question is clear, but I have not yet been able to carry out objective testing.

zylka5, 2024-02-09:
Nie tak dawno na częstotliwościach satcomu, na niedostosowanej antenie odbierałem słabe dźwięki syren było to na sofcje ijv przy właczonej funkcji satcom. Jako że sygnał był ciągły i stabilny postanowiłem wgrać kilka innych softów by sprawdzić czy na którymś uda się uzyskać jego odbiór, na żadnym się nie udało. Sygnał powrócił dopiero po ponownym wgraniu ijv. także fajnie jak by się taka lub lepsza  funkcja pojawiła :D

slavikby, 2024-02-09:
**[zylka5,** That's right, I myself was very surprised by the practical difference in reception, depending on the firmware!!!! At the moment, the leaders in reception sensitivity in the Satcom range are firmware from kamilsss655 and IJV. I think everyone will be very happy and grateful if the respected author makes improvements in reception in his firmware.

zylka5, 2024-02-10:
W jakiej sytuacji/przy jakich ustawieniach oprogramowanie kamilsss655 poprawia odbiór sat-com? Z chęcią przeprowadził bym testy, żeby zobaczyć czy jest różnica, między nim a stokowym egzumer :D

slavikby, 2024-02-10:
**[zylka5**, I’ll say right away that I don’t have any measuring instruments at the moment. Those. all conclusions are subjective and based on auditory perception. I received one of the Satcom frequency with the open squelch and changed the firmware several times in the same radio.

I will assume that it is not only the maximum gain that matters, but other parameters of the chip, and in particular, the reception bandwidth!