egzumer/uv-k5-firmware-custom #456: Signal TX no clear

marmatBCN, 2024-02-03:
I have been using Firmware Egzumer for a long time,
  However, there is a problem: when transmitting on the 2m and 70cm bands, a quiet parasitic tone can be heard in the background.
TX signal is not clear.
I downloaded the factory version and this problem does not exist, so it is clearly a matter of Exzumer firmware...

mentanah, 2024-02-03:
Can't confirm that. Using scrambler?

KvallVall, 2024-02-03:
Set backlight to 10 or off.

https://github.com/egzumer/uv-k5-firmware-custom/issues/405

marmatBCN, 2024-02-03:
I don't know what scrambler you're asking about?

marmatBCN, 2024-02-03:
Scrambler is off, backlight I tested

marmatBCN, 2024-02-03:
I set BKLight on 10 now is ok...

marmatBCN, 2024-02-03:
Yes thanks I set 10 now is okWysłane z MM iPhoneRegardsMarek SQ7BCNWiadomość

closed 2024-02-04 by egzumer