egzumer/uv-k5-firmware-custom #458: Stuck in F + 1 BAND mode?

MikeGriffinReborn, 2024-02-04:
I updated to the latest version V22 and tried the F +1 Band mode and now I cannot get out of that mode, turning the radio off? Nothing gets me out of this new mode?

MikeGriffinReborn, 2024-02-04:
I had to pull and restore my Chirp values to get out of that new mode, not sure what that was about, but won't be trying that anytime soon again !! lol

MikeGriffinReborn, 2024-02-04:
From the main page on the new release ...

Added bands switching to FM broadcast radio function. in FM radio mode press F + 1 BAND or long press 1 BAND. These bands are available in BK1080, and not defined by me. The main benefit is that you can set the band that is used in your country and auto scan function won't scan outside of it.

egzumer, 2024-02-04:
> Added bands switching to !!!`FM broadcast radio function`!!!

What you are doing you are copying memory settings to frequency mode, nothing funky about it

closed 2024-02-04 by egzumer