egzumer/uv-k5-firmware-custom #459: blacklist on scan

OE1MWW, 2024-02-05:
On scan, there is sometimes interference (kind of modulated tone) on a repeater channel - different timing and duration, but just strong enough to open the squelch.

It would be desirable to have a temporary 'black list', for example, if the scan stops on this 'interferer', that you press the 'F' button (or another ?) and from now on this channel is skipped during the scan.

AubsUK, 2024-02-21:
Memory channel scanning or frequency scanning? - I suspect channel, because you mention "repeater channel".

I'm looking at doing some things with memory scanning and this sounds like an interesting option, but that'll be done on a fork of Egzumer's.

MicaelJarniac, 2024-03-12:
On CHIRP there's a "Skip" column, but I have no idea if it's related to this at all.
[image]

zerog2k, 2024-08-29:
yes, coming from almost all other kinds of radios, the channel list thing seems like what others (e.g. Yaesu) consider to be scan "banks", but this radio seems to be missing a feature known as "skip", where a memory has a flag on the channel which allows you to increment through it, but it does not get tuned as part of channel scans.

I think this radio is missing this. There are frequencies that I want to have entered into the memory channel, but I don't want them to be part of any scan.
The scan "all" (`*`) is useful to scan _all_ channels, but it's not fully useful without a functional "skip" feature.

I'm studying the existing firmware operation of scanning, and may take this on as a side project to improve this functionality in the future. Because of the way the settings and channels are laid out in memory, and also how CHIRP interacts with the radio memories, It may be a significant departure in functionality, and be a new "fork", which may require "variant" support in CHIRP. (But that also shouldn't be too much of an issue.)

prokrypt, 2024-08-29:
Check out Armel's fork called F4HWN. Hold M when scan has paused on a station and it's blacklisted until next power-on.
I have my own fork of the firmware too with slightly different features, including the ability to enable monitor mode while in scan mode so you can scroll through the scan list.

AubsUK, 2024-08-29:
> I think this radio is missing this. There are frequencies that I want to have entered into the memory channel, but I don't want them to be part of any scan.
> The scan "all" (`*`) is useful to scan _all_ channels, but it's not fully useful without a functional "skip" feature.
> 

I've added 10 ScanLists and temporary/permanent lockout to my fork of this firmware which can be configured with the skip in CHIRP or in the radio. Each channel can be in one or more ScanList. When scanning, press the ScanList number to enable/disable. When on a broadcasting channel, press * to lockout until power cycle.

https://github.com/AubsUK/uv-k5-firmware-custom