egzumer/uv-k5-firmware-custom #463: Feature Request: Ability to change spectrum analyzer sweep speed

TechnoZombie, 2024-02-06:
Basically thats what it is stated in the topic.
I feel that, when using a wider range on the spectrum, it is too slow to sweep it.
It would be great if we could increase and decrease the sweep speed.
Not sure if this is possible, tho!
Thank you!

egzumer, 2024-02-06:
Yes it is possible, use narrower range to increase sweep speed, and use wider range decrease sweep speed.

Isn't that obvious that scanning more space takes longer time than scanning less?

closed 2024-02-06 by egzumer

TechnoZombie, 2024-02-06:
But that was my point! 
It would be nice to have a faster speed in wider ranges!
Almost like the speed in the memory channels scan.

egzumer, 2024-02-06:
It would be nice to have cars that burned 1L of gasoline capable of speeds around 800km/h.

TechnoZombie, 2024-02-06:
I get it, you're rude. Fine.