egzumer/uv-k5-firmware-custom #467: Stuttering audio on Narrow FM

Vinblastine, 2024-02-11:
I am using the latest firmware 0.22 on a UV-K5(8).

I have noticed on several occasions that when I am listening to a station on 2m and I have the radio set of NFM (which is correct for the UK band plan on 2m) that the audio stutters when the station is talking. The radio appears to loose the signal for a fraction of a second but it can happen in very quick succession (like stuttering)
When the audio cuts the green reception light goes out and the “RX” text on the displays cuts out too. 
I have noticed that it only seems to happen during the transmission of voice I.e. if the other stations is still transmitting but not speaking the issue does not occur. 

Has anyone else encountered this issue?

mentanah, 2024-02-11:
Happens the same, if squelch is totally open?

bsdinis1, 2024-02-11:
I've been experiencing the exact same issue, Squelch is set to 1 in the menu, when someone starts transmitting the radio opens the squelch but when they start speaking the radio closes the squelch, the squelch seems to be closing when there is loud audio coming through. I can listen just fine when I open the Squelch/Monitor.

egzumer, 2024-02-11:
It happens when you receive wide signal with narrow setting.

ye110wd, 2024-02-11:
It happens with any fm over a certain level of audio.

I thought it had something to do with audio clipping or a similar problem as AM had but setting squelch 0 keeps a solid green light even if there is audio clipping.

bsdinis1, 2024-02-11:
> It happens when you receive wide signal with narrow setting.

This was happening on a Wide FM repeater.

egzumer, 2024-02-11:
> This was happening on a Wide FM repeater.

That is what I said. If you receive Wide FM with radio set to narrow this is what you will get. I experienced it before with two handhelds. When two radios were close to each other it was ok, but from some distance the stuttering made communication impossible until the other radio was set to narrow too.

There is nothing I can do about it you set the BK chip in a certain way and it does it things. The squelch open/close is reported by the BK chip. The stuttering happens because this is what is reported.

thirtythreeforty, 2024-03-26:
I experienced this on NASCAR's MPN broadcast - even on the wide FM setting, squelch would continue to cut in and out. Going to monitor mode made the transmission sound fine.

Is it possible to further widen the receive bandwidth, maybe a new "very wide" bandwidth setting? The FM broadcast radio mode seems to do this, but I don't know if this is possible to set on ordinary VFO tuning.

AubsUK, 2024-03-26:
I have seen this regularly too. NFM repeater with NFM configured in the radio.

One thing I've done recently that I think might fix this, but I can't be certain, was to turn on (or off, I can't remember) this option:

`STE` - squelch tail eliminator, eliminates noise at the end of a transmission

thirtythreeforty, 2024-03-27:
For me, STE did not help; I tried both ways. I did also notice in my case that it appeared DTMF tones were being decoded, which I didn't expect, but I don't know if NASCAR legitimately sends those (the actual transmission settings are a little under-documented).

tomaz1, 2024-03-30:
I encounter the same issue with my device, specifically when operating with a Narrow FM (NFM) repeater, utilizing CTCSS. I've noticed that the audio reception is inconsistent, characterized by a **choppy,** where the sound seems to **cut in and out** abruptly. This occurs despite the signal strength indicator showing a reception level of, for example, S4. Intriguingly, when I switch the device to **monitoring mode**, or effectively disable the squelch by setting it to zero, the audio clarity is markedly improved, and the issue appears to be resolved.

Additionally, I have a Baofeng unit nearby that does not exhibit this problem under identical conditions, which suggests that the issue may be specific to the device/firmware or its settings rather than the signal or environment. Like thirtythreeforty mentioned, my device also unexpectedly displays DTMF codes on the screen during these occurrences, implying that it might be incorrectly processing DTMF tones instead of properly transmitting the audio.

Initially, I thought it was a squelch issue, but it seems unlikely to be a squelch problem if the signal is strong, or am I mistaken? I'll experiment with enabling the STE feature to see if it alleviates the issue. Does anyone have any other suggestions?

dadanielat, 2024-04-07:
I have the same problem after upgrading from 0.21 to 0.22
I did not notice it on 0.21.

tomaz1, 2024-05-02:
The STE feature didn't help. However, I installed the **armel** version from the link above and set the long press on the M key to "SWITCH WIDE NARROW." This seems to handle the audio cutting out much better. I'm wondering if the firmware could potentially detect whether the signal is wide or narrow within a few milliseconds and automatically switch accordingly?

selseyastro, 2024-07-24:
This is just on the peaks of modulation.  The SQ drops and maybe a short delay on the squelch line.  Maybe only a few miliseconds.   Or a menu item to set delay.  Just an idea.  I think it is some signals are over modulated on NFM.  Does not seem to happen on repeaters..

yesilug, 2024-08-20:
I'm having the same problem. 2 devices are version 4.00.7, I installed egzumer 0.22 on one of them. Both of them have the problem of stuttering audio. Have you found a solution?