egzumer/uv-k5-firmware-custom #475: No RX Audio on Simplex Channels

K4NHA, 2024-02-17:
Just dropped the latest firmware onto a new radio (Feb 16, 2024) and repeater memory channels work perfectly. Simplex channels however show a carrier on RX but no audio. Radio works fine on stock firmware but after flashing egzumer it has this issue on both manually programmed and CHiRP (latest) programmed channels. Anyone else ran into this issue?

egzumer, 2024-02-17:
You have to have RX CTCSS turned on, I don't see any other option. Show a video of what happens and what the LCD shows.

K4NHA, 2024-02-17:
https://github.com/egzumer/uv-k5-firmware-custom/assets/49967286/ac7dfc95-40d7-429a-84d9-3346778408f8

egzumer, 2024-02-17:
Turn off `D Decd`

closed 2024-02-17 by egzumer

K4NHA, 2024-02-17:
I knew I was missing something simple. Thanks for your help and sorry for the trouble!