egzumer/uv-k5-firmware-custom #482: Menu Key disable

EmCom1, 2024-02-20:
Is it possible to produce a version of egzumer v0.22 with the menu Key disabled for use by inexperienced volunteers who might inadvertantly, or otherwise, change settings?

ErikS-web, 2024-02-20:
The UVK5-serie is a HAM radio / walkie-talkie, with which radio amateurs enjoy the hobby

For the uneducated users, it is much easier to purchase radios, which can be operated to a limited extent because they have fewer buttons and only function on frequencies that you are allowed to use without a license.

OE1MWW, 2024-02-21:
@EmCom1 consider this - if you dealing with such uneducated and rough users, there are
more button function on this firmware, like 'activate VFO, A or B, + enter whatever frequency' etc. 
that could prevent those nosy users from diverting of the right communication channel or 
mode. The 'F' key followed by a number (an easy mistake!) brings functions that inexperienced users 
have never even dreamed of. Like spectrum analyzer etc. If they start to poke further more buttons, 
they will be completely lost. There would be no difference if those users activate the 'Menu' button or 
poking on the 'F' + following by another button and change the designated functions of the UV-K5.

Our local group, had similar demand, we equipped a group of users with devices that fulfil exactly 
those requirements: no unscrewable antenna,  just a volume control, channel button (voice announced 
channel name in english) and ptt buton. Settings are done by software, direct channels, repeater 
offset, tone control, etc., (Aliexpress, device name KD-C1)

The Chinese dealers on Aliexpress offer a variety and cheap devices with such funtionality.

EmCom1, 2024-02-21:
Thanks OE1MWW - we have sets of such "simple locked down" radios for "uneducated users" for events etc. and as you say they work well in those situations.

The issue we have is complex. We have as you suggest disabled VFO permanently using the CPS software  and this prevents F+3 activating VFO mode etc.etc. and are happy with the keyboard functions still available. It is not rough users - we will train them and they are volunteers with other legitimate organisations - it is the ability to inadvertent change some parameters (particularly TX frequency) from the keyboard that would prevent the UVK5 complying with some of our OFCOM Private Business Radio (PBR) licence requirements. 

The particular most important feature of the UV K5 is that we are able to programme some memories for RX only whilst others remain RX and TX. The radio therefore serves two purposes (i) a versatile RX only for Broadcast FM, amateur, marine etc bands but also (ii) allows TX/RX on our licenced PBR frequencies and is interoperable with other organisations for which we hold licences such as the Red Cross. St John Ambulance etc.

**We are happy with the functions currently available from the keyboard - we just need to disable the Menu Key to make it comply with the various licences we hold.**

ErikS-web, 2024-02-21:
If you first read the Online Manual (Wiki) completely. Egzumer and I have certainly put so much effort into this.

Then you come to the conclusion that it is already possible with [F # 🗝](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Button-functions#f--)
That blocks All keys on the Front panel.
Only the side keys still work and you can also disable them individually or give them a specific function.
All the operator has to do is stay away from the F ;-)

We try to get the _most out of the radio_ and all that functionality must fit into **60kB.** 
Wasting memory to _further limit functionality_ in EGZUMER image seems like taking water to the sea.

That doesn't stop you from creating unlimited firmware yourself, as this is alle open source 👍

egzumer, 2024-02-21:
@ErikS-web can you stop moralizing? Please if want to help, help by giving facts, not by sending people to the wiki when it is clearly irrelevant.

It is certenly possible to make the quansheng a one button HT for dummies, it is not that hard and it does not take that much space. The only issue is that I have to spend time on a feature for just a small portion of users, that I myself will never use.

EmCom1, 2024-02-21:
Firstly I would like to thank everyone for all the hard work they have put in on developing this firmware for the UV K5 – it is very much appreciated.  I am heading up our EmCom Project as a trustee of the charity but I do not have the skills, or unfortunately the time given the other commitments to the project, to learn how to code the firmware from scratch.

I also fully understand the issues regarding valuable memory space.

Perhaps I have not explained properly that I do not want to disable all the keys on the front panel. The up/down and number keys are required for changing to another memory/channel and there is no problem regarding the other functions available – we have already disabled the VFO to comply with that part of our PBR licence.
It is only the MENU key I would like to disable. I had hoped there might already be a simple flashable patch that would overwrite some code and disable the MENU key.

I also fully understand spending time on a feature for just a small portion of users is a big ask. If the EmCom Project Pilot is a success it will first involve up to 30 voluntary and faith sector volunteer groups and a couple of hundred radios for their leader teams within our county. It has the potential to go nationwide. 

Am I correct in thinking that a flashable patch would be feasible to disable the MENU Key? -  and perhaps applicable to future versions of the firmware without having to be rewritten?

Thanks for taking an interest in this Project

egzumer, 2024-02-21:
If I were to do it, it would be chirp switch  similar to "frequency mode allowed".

EmCom1, 2024-02-21:
It would be great if you could - I will research what a "chirp swith" is - but I guess it is something like what I used to disable VFO mode using the Quansheng CPS. 
P.S. The Civil Air Support Group say thanks for the AM Mod it has made all the difference.

emanuelegissi, 2024-11-03:
Disabling the menu button would be a great feature for us as well. We are firefighters...
Thank you for the great work.

emanuelegissi, 2024-11-03:
> have a look here and may be ask the dev he's pretty good and adding more
> and more features all the time

Thank you, I am going to try!

Orcuthe, 2025-03-21:
Did anyone solve this? I want to lock down some UV-K5 handsets to give to children to use and this would be a beat solution.

(Ideally I'd like to limit menu access to a number of menus - eg VOX, VOL, VOICE, BEEP, ABR, SQL)

I wonder if AI could help an amateur create this firmware version...

ldicdican, 2025-03-21:
Toggling menu items you want to hide to zero("0") maybe?

Orcuthe, 2025-03-21:
> Toggling menu items you want to hide to zero("0") maybe?
> 
> Or just remove the item here? ui/menu.h
> 
> Sorry Armel and you guys, just guessing though.
> 
> Let's just wait for Armel to direct us where and what is the right path to do.
> 
> Btw, soooo love this firmware! Great job Armel.

Yeah, being able to select which additional menus could be added to the "hidden" list would be perfect. Is that what "toggling menu items you want to hide to zero" does? Where do you do this? Can those menus still be accessed if you know the right key press?

Sorry for all the questions, but it looks like you've just solved my problem if I understand it all correctly!

ldicdican, 2025-03-21:
> Yeah, being able to select which additional menus could be added to the "hidden" list would be perfect. Is that what "toggling menu items you want to hide to zero" does? Where do you do this? Can [...]

don't have much time for now to test that idea, but I think it's the best thing to do, if you have time then you should try it remove the item from the menu list:

const t_menu_item MenuList[] =
{
//   text,     voice ID,                               menu ID
	{"Step",   VOICE_ID_FREQUENCY_STEP,                MENU_STEP          },
	{"TxPwr",  VOICE_ID_POWER,                         MENU_TXP           }, // was "TXP"
	{"RxDCS",  VOICE_ID_DCS,                           MENU_R_DCS         }, // was "R_DCS"
	{"RxCTCS", VOICE_ID_CTCSS,                         MENU_R_CTCS        }, // was "R_CTCS"
      .....
      .....

the whole row of the menu item should be removed, like this one:
{"TxPwr",  VOICE_ID_POWER,                         MENU_TXP           }, // was "TXP"

should be like this after removing:
const t_menu_item MenuList[] =
{
//   text,     voice ID,                               menu ID
	{"Step",   VOICE_ID_FREQUENCY_STEP,                MENU_STEP          },
	{"RxDCS",  VOICE_ID_DCS,                           MENU_R_DCS         }, // was "R_DCS"
	{"RxCTCS", VOICE_ID_CTCSS,                         MENU_R_CTCS        }, // was "R_CTCS"
      .....

Orcuthe, 2025-03-21:
> don't have much time for now to test that idea, but I think it's the best thing to do, if you have time then you should try it remove the item from the menu list:
>
> const t_menu_item MenuList[] = { [...]

Thanks a lot for that. I'll have a think.

I've been looking at another approach that might work better for my needs. Instead of completely removing menu items from the firmware, I'm interested in moving certain menu items to the 'hidden' group.

Looking at the firmware code in main.c, there appears to be a mechanism already in place for this:

```c
// count the number of menu items
gMenuListCount = 0;
while (MenuList[gMenuListCount].name[0] != '\0') {
    if(!gF_LOCK && MenuList[gMenuListCount].menu_id == FIRST_HIDDEN_MENU_ITEM)
        break;

    gMenuListCount++;
}
```

This code shows that each menu item has a 'menu_id', and there's a constant called 'FIRST_HIDDEN_MENU_ITEM' that marks where hidden menus start. When gF_LOCK is false (normal operation), the radio only shows menu items with menu_id less than FIRST_HIDDEN_MENU_ITEM.

Instead of removing menu entries, I'm thinking of:

1. Finding where FIRST_HIDDEN_MENU_ITEM is defined (probably in ui/menu.h or menu.c)
2. Ensuring the menus I want to hide have menu_id values equal to or greater than FIRST_HIDDEN_MENU_ITEM
3. Possibly reordering the MenuList array to group hidden items together

This would keep the menus in the firmware but hide them from normal access. The special boot combination (which sets gF_LOCK=true) would still reveal all menus when needed.

Does this approach make sense? Do you know where FIRST_HIDDEN_MENU_ITEM is defined or which menu_id values typically correspond to hidden menus?

ldicdican, 2025-03-21:
That's the BEST approach I think right now, better than my approach I believe.

Try it and we'll see, best of luck 👍

Orcuthe, 2025-03-22:
I'm waiting for the handsets to arrive to test, but after more investigating, I *think* this might be as simple as editing ui/menu.c to move any menu items I want to hide so they appear after  the item with menu_id == MENU_F_LOCK

@emanuelegissi and @EmCom1 - what did you do in the end? I think my solution might work for you both too.