egzumer/uv-k5-firmware-custom #484: Poppyseed display and ridiculous battery voltage

jholubinka, 2024-02-20:
First of all, sorry for the stupid title.

I'm reporting the issue for version 0.22, but it's much older than that. In fact, the last version which worked for me was 0.15 and it's present since 0.16. I'm reporting this right now, because recently I didn't have time to play with modding my radio. Sorry for that as well.

**Hardware**: UV-K5(8). I don't know the exact h/w revision, but I bought the radio almost immediately after it hit the market.
**Steps to reproduce**: Flash firmware to 0.22, disconnect the cable & reboot.
**Expected behaviour**: Radio shows splash screen and boots up
**Actual outcome**: A screen full of "poppyseeds" - multiple pixels in random places, no backlight, no boot-up.

[image]

However, when switching on the device is connected either via K plug to the computer or via USB-C to a charger, **my radio boots up properly, but in both cases it reports wrong battery level** (0% and 0.24V) and therefore cuts off TX. It's **not** a battery issue, because:
1. After booting up the cable may be unplugged and the radio still works, even with such crazy voltage
2. The radio works properly (and reports good battery level) on stock firmware and other mods that are not forked from this one.

Unfortunately I don't have any more UV-K5(8)'s to test it, but I guess that it's a rare issue (maybe my radio is somehow faulty), since I couldn't find any similar problems. That's why I'm not expecting this issue to be solved, but I'm logging it anyway, so it can be googled in the future if the problem turns out to be more common.

Kind regards,
Kuba SP9VIK

egzumer, 2024-02-20:
You flashed fagci reborn before, didn't you?

jholubinka, 2024-02-21:
As a matter of fact, yes, I did! :) Thanks for pointing that out, that put me on the right track.

I've solved the issue by overwriting original eeprom I've found in @sq5bpf's k5prog repository using "Write Configuration" command from @OneOfEleven's k5prog-win (and ignoring the warning about file size mismatch). Now the voltage reported by the radio and the actual voltage measured from the battery differs only in 0.02V, but that's a difference I'm willing to accept ;)

I guess that writing proper calibration data would solve the issue, but unfortunately I've lost my backups.

egzumer, 2024-02-21:
Chirp configuration image contains full epr

joaodezecha, 2024-02-24:
Kuba SP9VIK
Tenho o mesmo problema quando atualizo para versão  firmware para 0,22
       PU8PCF

joaodezecha, 2024-02-24:
Kuba SP9VIK
I have the same problem when I update firmware version to 0.22
PU8PCF