egzumer/uv-k5-firmware-custom #492: [Attempted tx out of set parameters] With duplex Explicitly set to "off" and offset blank in cherp. TxODir sets self to "-" and TxOffs to Current channel frequency!

The-One-Who-Knoks, 2024-02-25:
I've been wondering why i kept getting Tx Disabled, despite enabling default+, and I just found why. 

I was starting to think my license frequency's (BRSL) were not covered for some reason under default+. After scratching my head for a bit and wondering if I borked the flash, Or misunderstood some of the settings. I went through them one by one on both radios and noticed TxODir was set to "-" and TxOffs Set to the frequency of the channel...

I went back into chirp and saw I explicitly turned duplex off as it is not allowed on my license. After setting the duplex field blank again on all channels and resending the config to my Retevis RA79, TxODir Is correctly set to "off" and TxOffs set to "0.00000"

Just to confirm, I explicitly set Duplex to off in chirp again with the Offset blank, and the same behavior happened. Duplex enables itself to "-" And the offset is set to the channel frequency.

ErikS-web, 2024-02-25:
Check if modulation is FM

The-One-Who-Knoks, 2024-02-25:
It is on FM. The Radios are working as expected by leaving the duplex field in CHIRP blank.
The main question is, why explicitly setting it to off Is causing the radio to set  TxODir to "-" and TxOffs to the channel frequency...

mentanah, 2024-02-25:
Dumb question, using latest custom chirp driver from egzumer?

egzumer, 2024-02-25:
Duplex "off" doesn't mean it turns duplex off. It means TX disable. Don't ask me why, this is chirp logic. It sets the radio tx freq to 0MHz effectively disabling it.

closed 2024-02-25 by egzumer

The-One-Who-Knoks, 2024-02-26:
Thanks for the explanation. 👍 On the bright side, I now know how to turn TX off on a set channel.