egzumer/uv-k5-firmware-custom #496: TX Offset = memory channel on power cycle

disgusdad, 2024-03-06:
EGZUMER 0.22 release firmware. With TX all unlocked every time I turn the radio off then back on, all memory channels have there TX Offset values set to the frequency of the channel. I can enter the menu, set the tx offset to 0.0000 and it will work until I cycle power on the radio again.

ErikS-web, 2024-03-07:
Did you read this #492

egzumer, 2024-05-19:
It looks like you set duplex to off, which in chirp is a way of disabling TX for a channel. In QS this is realized in a "hacky" way by setting the offset so TX freq is on 0MHz.

D33zN0tz84, 2024-06-09:
I Have the same bug in the firmware, when you unlock all with the secret menu. 
The 27mhz's offset is set to the frequency of the cb chanel you put it on. 

So when you want to transmit it shows 018.0000 on the screen and not 1 of the 40 cb chanels you have chosen. also you cab bot change and store the offset. normaly i set it on 0.00000 TX MHZ is the same as RX MHZ.

Example CB chanel 19 027.18500mhz  shown on the screen to listen, if you transmit it shows 0.180000mhz
If you go into the menu you see that offset is set to 027.18500, you can change thos to 000.000 and save but after changing chanel and returning to chanel 19 the offset is again changed to 027.18500

Or am Imissing a setting to adjust this?

Sorry fot my english , this is not my native language.

thanks :)

egzumer, 2024-06-09:
Changing settings in channel mode has temporary effect, unless you save the channel settings.

ps. you should not transmit on CB channels with this radio.

closed 2024-06-09 by egzumer