egzumer/uv-k5-firmware-custom #497: Fix TX sensitivity

Valinstein, 2024-03-06:
On egzumer 0.22 tx sensitivity on UHF -119db, but on stock was 126-127db. Would be nice if in the next relises sensitivity come back to stock firmware results

OE1MWW, 2024-03-07:
@Valinstein. Did you mean 'RX sensivity' (?). How did you measure ? Squelch open ? What carrier- plain or FM modulated carrier? Btw., sensivity is expressed 'dBm' and not in 'db'. The average sensitivity of a UHF receiver is typically around -110 dBm to -115 dBm. So, your shown values (if they are in dBm) are not so bad - consider this - for a device, delivered to your front door for 20$.

OE1MWW, 2024-03-07:
Checked with a 
UV-K5 on egzumer 0.22
UV-K5(8) on kamilsss655 firmware
Signal generator: tinySA + 40db attenuator, FM modulation 1kHz 3Khz dev.

On both devices:
VHF: -130dBm (-90dBm +40 db)
UHF: -125 dBm  (-85dBm + 40 db)

Valinstein, 2024-03-07:
Here you some examples:
Egzumer:
https://youtu.be/xpkcRJWbqCw
Stock firmware:
https://youtu.be/nfLkkS8Ds8Q

kamilsss655, 2024-03-07:
Hey guys, I don't want to be rude, but this is completely wrong.

You are not measuring receiver sensitivity:
> The sensitivity of an electronic device, such as a [communications system](https://en.wikipedia.org/wiki/Communications_system) receiver, or detection device, such as a [PIN diode](https://en.wikipedia.org/wiki/PIN_diode), is the minimum [magnitude](https://en.wikipedia.org/wiki/Magnitude_(mathematics)) of input [signal](https://en.wikipedia.org/wiki/Signalling_(telecommunication)) required to produce a specified output signal having a specified [signal-to-noise ratio](https://en.wikipedia.org/wiki/Signal-to-noise_ratio), or other specified criteria.

>Receiver sensitivity is a measure of the minimum signal strength that a receiver can detect. It tells us the weakest signal that a receiver will be able to identify and process.

What you have measured is *squelch sensitivity*, and it can be increased so its always open, but that would not be very practical. 

Also please note that the dBm values are just numbers and can have different meanings across firmwares based on their calibrations.

Valinstein, 2024-03-07:
Even with open squelch it doesn't hear anything on levels above 120dbm...

Ok, i tried. If  this is not issue but _feature_ - ok...

OE1MWW, 2024-03-07:
@kamilsss655 (my prefered firmware designer ;-) )
dBm is 1 milliwatt on 50 ohms as a reference. All the dBm values discussed here are the ones on the generator side.
-120 dBm in a 50 ohm system is approximately 2,449.49 microvolts (µV) input.

BruMcD, 2024-03-12:
So much missinformation

The on screen dBm figure is an estimate at best.  It certainly isn't accurate on mine

FM Sensitivity is measured by looking for a condition called 12dB SINAD.   This means the wanted signal (a 1KHz tone at usually 60% of max deviation ) is standing 12dB above the surrounding noise level.   After a few years you can often judge this by ear but otherwise and for Precision you need a SINAD meter/speaker load and a generator that doesn't leak RF and can accurately put out a precise signal and defined carrier level.

Other ways to define 'sensitivity' exist- like squelch opening or noise quieting - but they are NOT the way the specification calls for and are inferior to SINAD in FM systems.

In doing these tests you realise Squelch must be open and that 'better' figures will result from using a narrower bandwidth NFM receive filter with the generator set to wide FM (5KHz max dev) as the noise bandwidth is less and the tone is now 100%+deviation .    Also any difference in tuning accuracy between Generator and Radio will degrade the SINAD.    Any noise on the generator or spurious noise on the test channel will impact measurements too 

Across several sets all using 0.22 and the same CHIRP settings I've mostly seen 12dB SINAD sensitivity figures of -121 to -123dBm  
Which I'd call close enough to the specified -123dBm  dBm for a Chinese set :-)
The difference presumably relates to hardware issues like Chip quality, internal noise and LPF losses

BruMcD, 2024-03-12:
> @kamilsss655 (my prefered firmware designer ;-) ) dBm is 1 milliwatt on 50 ohms as a reference. All the dBm values discussed here are the ones on the generator side. -120 dBm in a 50 ohm system is approximately 2,44 microvolts (µV) input.

hmmmmmmm    Maybe he should stick to Firmware. 
-100dBm is 2.**2**4uV PD at the antenna terminal       -120dBm is 0.224 uV PD

OE1MWW, 2024-03-12:
@BruMcD ... you got me. First I had 6 digits behind comma, then I edited my own post ... late evening typo - set the comma wrong. So, I owe you some dBm ;-)