egzumer/uv-k5-firmware-custom #502: Can no longer control scan lists ...

MikeGriffinReborn, 2024-03-14:
I cannot control the scan lists anymore. According to this video (options are a bit different but still the same) I should be able to turn off, or add to list1 or list2 for any channel, see => https://www.youtube.com/watch?v=oGIJJI0cX4w

I cannot remove a channel from the scan lists, so I have to remove things like the Weather service that are always broadcasting from my radio.

MikeGriffinReborn, 2024-03-14:
Even though my Weather frequency has both ScAdd1 and ScAdd2 both set to OFF it comes up in the Scan everytime

MikeGriffinReborn, 2024-03-14:
My observation are thus:

1) Any frequency not in a scan list somehow is scanned in Scan List 1 always
2) If you turn on the radio, press scan, scan list 1 is scanned, if you hit exist, then scan again, scan list 2 is scanned, this is great, however, if you hit exit and scan again, it's always scan list 2, you have to turn off the radio to get to scan list 1 again.

And again, the frustrating part is anything in marked as off in both 1 and 2 is always scanned in 1.

Apparently the Quansheng has a major flaw, it always scans All channels on one of the scans. I have removed the weather frequencies from the radio, bummer but at least I can scan.

They screwed up, the BaoFeng has  a simple skip flag, so should the Quansheng, the all frequencies scan is always going to stop on a weather service, thus, apparently you cannot have those on Quanshengs

egzumer, 2024-03-17:
https://github.com/egzumer/uv-k5-firmware-custom/wiki/Radio-operation#memory-channels-scanning

AubsUK, 2024-03-19:
> They screwed up, the BaoFeng has a simple skip flag, so should the Quansheng, the all frequencies scan is always going to stop on a weather service, thus, apparently you cannot have those on Quanshengs

When you're scanning, press and hold `*` - It should cycle between LIST1/LIST2/ALL.

I've got a fork of Egzumer's firmware https://github.com/AubsUK/uv-k5-firmware-custom/ where I've created 10 lists.  You can put any channel into one or more of the lists.  The downside, there is no "scan all", but I suppose you could put all channels into List0 and scan that.  I've also put in an 'always skip' and 'skip until power cycle'.
Afraid I have no knowledge on NOAA or if it works on my fork (it should) because we don't get it in the UK.

closed 2024-03-19 by egzumer