egzumer/uv-k5-firmware-custom #515: Is possible to merge with https://github.com/phdlee/uvk5cec ?

sveco86, 2024-03-21:
Is possible to merge with https://github.com/phdlee/uvk5cec ?

It implements nice spectrum analyzer / waterfall, please see videos here:

http://www.hamskey.com/2024/03/introducing-uv-k5-version-02v-spectrum.html

ErikS-web, 2024-03-22:
I don't think so, because there is not enough free space.
Latest image is above 98%

[image]

What you can consider is to run it on your PC.

This is possible with [Q-dock](https://github.com/nicsure/QuanshengDock/wiki).... a variant based on Egzumer.

[image]

josephburt, 2024-03-22:
> Is possible to merge with https://github.com/phdlee/uvk5cec ?
> 
> It implements nice spectrum analyzer / waterfall, please see videos here:
> 
> http://www.hamskey.com/2024/03/introducing-uv-k5-version-02v-spectrum.html

I would not consider this an option as the https://github.com/phdlee/uvk5cec is a firmware based on modified K5 hardware 
with the SI4732-A10