egzumer/uv-k5-firmware-custom #516: RX problem 5+ seconds ON

papatakhs, 2024-03-22:
why is it slow to download to the rx, the save is off to sql 1 (when it downloads to the above rx then it needs a few seconds to start downloading) with other Firmware it is ok[

https://github.com/egzumer/uv-k5-firmware-custom/assets/164517623/f6939ec2-9298-404d-98e0-af2e58513ee8

ErikS-web, 2024-03-22:
Please explain more expliciet. 

I do not understand, what you are trying to demonstrate

papatakhs, 2024-03-22:
he is slow to do rx 
 59. `RxMode DUAL RX RESPOND

josephburt, 2024-03-22:
So what they are showing is when in the 446 its quick to respond. He changes his second radio to the other band 142.200 and the K5 is still monitoring 446 when he keys his second radio it is slow to switch to the second channel to hear the rx on 142.200.

ErikS-web, 2024-03-22:
Here is some original text I wrote for the wiki. 
It explains this functionality in some more detail.
Especially the last line ...
[image]

josephburt, 2024-03-22:
Thanks for the information @ErikS-web  it makes sense to me and I appreciate the documentation.

egzumer, 2024-03-22:
Dual watch stays on the last received frequency for a couple of seconds, this is expected behavior.

closed 2024-03-22 by egzumer