egzumer/uv-k5-firmware-custom #521: DTMF call partially working if stock firmware calls EGZUMER o EGZUMER calls stock firmware.

Efiartes, 2024-03-24:
I have two quansheng uv-k6, if I configure D-RSP option to REPLY or BOTH and make a DTMF call between them with stock firmware it works without problem.
If I load EGZUMER in both walkies it also works without issues.
If I mix one walkie with stock firmware and the other with EGZUMER DTMF call works but reply is not detected by the originating side of the call, both walkies can communicate between themselves yet reply confirmation functionality is lost.

BruMcD, 2024-03-25:
Not enough facts / tests here to be sure but the usual perils are insufficient Delay after carrier up for the receiver to come out of power save and be ready to decode.  Likewise, at the called end, enough delay before the acknowledgment/reply that the sender is back on receive.  It can be real pain to get it working at the best of times (not just on Quanshengs !)

ErikS-web, 2024-03-26:
There where several DTMF corrections done in previous releases.

See https://github.com/egzumer/uv-k5-firmware-custom/releases?q=dtmf&expanded=true

Special look at release 0.17. So I think is that original firmware was NOT standard DTMF