egzumer/uv-k5-firmware-custom #530: Tx disable/ enable shortcut

rpifuntime, 2024-03-28:
Great firmware thank you. 

Id like to see a feature added with a shortcut to disable/ enable tx quicker, rather than having to switch off the radio and go into the secret menu. 
this radio goes everywhere with me and unless i disable tx i end up getting ptt wedged against something.

AubsUK, 2024-03-28:
Wouldn't it be sufficient to press and hold `F` which enables/disables the key lock?

rpifuntime, 2024-03-28:
Yes unfortunately the f key doesn’t lock ptt, would be handy to have either the f key lock ptt or maybe another method of just toggling it on and off without changing any of the settings in the secret menu.

ErikS-web, 2024-03-29:
I'd say turn the radio OFF.  But I think you still want to continue listening on the frequency (where you are allowed to broadcast).

Then a solution could be to include that frequency EXTRA in your channel list. And then set as described here
https://github.com/egzumer/uv-k5-firmware-custom/issues/492

Mertszr02, 2024-04-06:
Yes! it's kinda easy to accidently press the ptt when handling the radio. Sometimes I just want to listen and not worry about disturbing anyone so any shortcut to enable/disable TX would be great.

rpifuntime, 2024-04-07:
Exactly, i tune mine into a node and listen in on nets while im doing things, some of the nets are 2+ hours long and i only speak for a couple of minutes at a time, you never know when they are going to call you either so not enough time to go to the menu and reactivate tx. would be such a handy feature

There are buttons that can be programmed but tbh i dont use them as the radio does everything else i want already from the shortcuts on the keypad, would be handy to be able to set one of the programmable buttons to disable/ enable tx

D33zN0tz84, 2024-06-09:
I think i found a bug in the firmware, when you unlock all with the secret menu. 
The 27mhz's offset is set to the frequency of the cb chanel you put it on. 

So when you want to transmit it shows 018.0000 on the screen and not 1 of the 40 cb chanels you have chosen. also you cab bot change and store the offset. normaly i set it on 0.00000 TX MHZ is the same as RX MHZ.

Example CB chanel 19 027.18500mhz  shown on the screen to listen, if you transmit it shows 0.180000mhz
If you go into the menu you see that offset is set to 027.18500, you can change thos to 000.000 and save but after changing chanel and returning to chanel 19 the offset is again changed to 027.18500

Or am Imissing a setting to adjust this?

Sorry fot my english , this is not my native language.

thanks :)

AubsUK, 2024-06-09:
If you try and transmit on 27 MHz, you'll only get milliwatts out at that frequency, but you'll get harmonics at higher power at lots of other higher frequencies, some of which might include air band and emergency services.  Strongly recommend not trying to transmit at 27 MHz using this radio.