egzumer/uv-k5-firmware-custom #536: Main Only RX, make the display more simple. (only show one VFO)

SutliffeProductions, 2024-04-01:
Hi Guys, this firmware is great. My only problem is complexity for family use, plus my eyes are not that good. Personally, I run my radio solely in main only RX mode i don't really ever use the second VFO. therefore i would like if i could have only one VFO displayed at a time in this mode (Or another mode, i don't mind) still being able to switch between a and b could be nice, however if the main thing is that if only 1 VFO is displayed, the main font could be larger, and easier on my eyes.  

Thanks guys,
let me know if I'm completely wrong and this is not a valid request.

jonnycastaway, 2024-04-02:
I also want such feature.

Akira25, 2024-04-02:
related to (or duplicate of) #124

closed 2024-04-02 by egzumer