egzumer/uv-k5-firmware-custom #538: Remote kill seems broken

Efiartes, 2024-04-03:
Remote kill seems broken, nothing happens when "Kill code" is sent, steps to reproduce:

1. Enable "Permit remote kill" option using Chirp.
2. Configure "Kill code" option.
3. Configure "Revive code" option.
4. Adjust two wealkies on the same channel.
5. Enable "D Decd" option in both walkies.
6. In one walki send the other side "Kill code".

In the other hand if you enable "DTMF kill lock" the walkie boots up locked, it receives radio signal(green led), no audio and can't transmit.
If I send from another walki the "Revive code" it receives the DTMF code, I can see it on display but there's no audio.
It seems that "DTMF kill lock" option completely blocks and disables the walkie, does it have any use case?

Best regards.

ascarons22, 2024-04-03:
Same problem, any solution? Thank you

BruMcD, 2024-04-03:
Their terminology is non-standard but Probably covers a few parts of the usual Stun / Kill functionality .   

A Stun **_(but here called Kill)_** command would normally 'silence' a radio and disable most controls until a DTMF Revive code is received from another radio.   Use case : Mislaid Radio

A Kill command  **_normally_** means it stays 'more than just Stunned with no radio or DTMF functions at all ' (So there can be no DTMF Un-Kill) until , if found'  it must be re-programmed at base or in some cases a further unlock code is entered on the keypad 
Use case: Lost or Stolen Radio  

So, They have used non standard  terms /process for whatever reasons    

But in this case it appears a stunned (aka killed) radio **with Kill Lock enabled** stays part operational yet not revivable or re-usable until re programmed (or perhaps even flashed and reset)   Use case :  Lost or Stolen radio  

From experience this stuff has minimal security, several flaws and no use case that will not cause a pile of aggravation :-)

closed 2024-04-10 by Efiartes