egzumer/uv-k5-firmware-custom #541: Unable to get rif of "TX DISABLE" using the "Unlock ALL" menu

jujumo, 2024-04-04:
I rebuild firmware on main branch (7607f0a4bd6203d1f06b70556fc1ce0d7399d6b3)
I followed instructions to "unlock all".
setup frequency "143.9875" Mhz and push PTT
Still get the "TX DISABLE" message.

I modified the code to lower the counter from 10 to 1 to ease debug (kind of annoying 10x for debug): still get the same result.

Is there something I am not doing properly ?

egzumer, 2024-04-05:
143MHz is available in normal operation, no need to unlock all, you are probably on AM.

jujumo, 2024-04-06:
Thank you for you answer. You are write, that was it. I even don't know I could switch that, maybe because its not the default behavior on other firmware. I hope this ticket can help other people that are confused, like i was:

If the "AM" logo appears just beneath frequency
go to "menu" > "demodu" (13) > "FM"

Thanks again, and sorry for the naive question.

closed 2024-04-06 by jujumo

2e0fth, 2025-04-06:
Unable to take radio off tx disabled when in am or ssb mode