egzumer/uv-k5-firmware-custom #546: Will there be a new release or the project is stopped?

T1000-Cyberd, 2024-04-15:
Will there be a new release or the project is stopped?

PalinuroSec, 2024-05-03:
up

mentanah, 2024-05-06:
Up

askondro, 2024-05-06:
Seems like that the flash memory is the main factor this is not advancing. I'm just guessing.

AsciiWolf, 2024-05-07:
@egzumer Please, consider tagging a new (0.23) release. (Maybe after fixing #552 if possible.) The last one was more than 3 months ago.

egzumer, 2024-05-07:
I'm out of time these days. I looked a bit at that issue some time ago, but it is a little complicated. I don't have anything new to release at the moment, and I don't know when I will, sorry.

ps. Thanks for the tip.

chrisdebian, 2024-07-22:
Hi, all.

I'm new to this project, but continually asking for updates, is considered a poor behaviour in the open-source world. The fact that people are asking though, is indicative that something may need to be looked at.

It's understandable that people want updates, but it's also understandable that egzumer has other work/ life pressures. 

Can I suggest the following?

1. egzumer release regular (weekly?) 'Beta' buids. These will have the caveat that they are not properly tested, and to encourage feedback from the community. For radio users who don't like Betas, they can wait for when @egzumer is able to release another stable release.
2. If the above isn't acceptable to the users of the software, then it is common for software to be forked and released separately. This isn't ideal, but it's an option.

What does anyone think?

Thanks,

Chris.