egzumer/uv-k5-firmware-custom #549: Loud pop through headset when squelch opens

loncaric, 2024-04-16:
On v0.22 when I plug in an external headset each transmission is proceeded by a loud pop when squelch is broken.

salokinrekced, 2024-05-02:
I believe this is more to do with a hardware issue of the PA turning on and off. UV5R has a similar issue.

egzumer, 2024-05-19:
Yes that is the case. It is especially due to external speaker being connected to one side of the PA differential output. The other side of the speaker is connected to ground. When PA is activated the outputs are switched from 0V to 1/2 supply voltage. That is where the pop comes from. There was a suggestion to keep PA active all the time, but this would greatly rise power consumption.

There is nothing I can do about that.

closed 2024-05-19 by egzumer