egzumer/uv-k5-firmware-custom #552: DTMF A isn't working.

fscii, 2024-04-24:
I was setting up HamVOIP and reached the DTMF testing stage of Asterisk.  All DTMF codes work as expected except the letter A.

I saw issue 85 (closed) where something went wrong with 6.
https://github.com/egzumer/uv-k5-firmware-custom/issues/85

It was suggested to try the DTMF decoding app:
https://play.google.com/store/apps/details?id=com.arcoirislabs.dtmfencdec.app&pcampaignid=web_share

The app successfully decoded all DTMF codes except A.

Radio:  Quansheng UV-K5
Firmware: 2.1.27

Behavior was repeated on a 2nd Quansheng UV-K5

egzumer, 2024-05-19:
You say "Firmware: 2.1.27". What do you mean? This is the original stock firmware not my firmware, in that case I have nothing to do with this.
DTMF codes are all verified and are exactly what they should be like. Last issue fixed regarding DTMFs was that some/all codes were transmitted too quiet.

@AsciiWolf you mentioned this issue. Initially I confused it with something else. What problem do you have here?

AsciiWolf, 2024-05-19:
Oops! I did not notice what firmware version the author of this ticket mentioned. I personally did not test this, but if it does not happen with the Egzumer firmware, there is nothing to fix and this ticket can be closed and/or ignored. :-)

fscii, 2024-05-19:
My bad on the firmware as I had a 2nd radio here at the time.  I have firmware EGZUMER 502300b.
I'm using the radio to interface with Asterisk 1.4.23 (running under HamVOIP image).  It decodes every DTMF sent but cannot pick up the 'A'.  I go to the cmdline interface and try every DTMF code and only 'A' never gets picked up.

egzumer, 2024-05-19:
I did all the tests before to make sure that it is all hunky dory with DTMF tones so I don't see what may be the issue. Read through #85. There is a description of all steps to diagnose what is the issue. Use other radio that works and compare what is the difference. You can also record the audio and share here so I can compare.