egzumer/uv-k5-firmware-custom #553: FM Radio does not start if squelch open

reald, 2024-04-26:
**Steps to reproduce:**
1. Set squelch to 0. Radio will receive something on current frequency.
2. Long press 0 for starting FM radio

**Expected behavior**
-  Radio will change to FM mode

**Observed behavior**
- Radio shows "FM" on the top left of the screen
- Radio does not change to FM mode
- Radio does not respond to keys except Exit

**Workaround**
1. Increase squelch until nothing is received
2. Hold 0, FM Radio starts normally

Observed with version 0.22.

Thanks and keep up the good work!

nikant, 2024-04-26:
I believe this is normal behavior

RX supersedes FM radio. As long as there is reception FM radio will not start. 
The symbol that appears indicates that IF RX stops at some point (which is impossible since squelch is 0) then the device will automatically will switch to FM radio (which will not happen as I wrote since SQL is 0)

egzumer, 2024-05-19:
It is how it works. When radio is in FM mode it still monitors the main frequency (dual watch doesn't work in FM mode by the way). It temporarily switches audio to main freq if there is a transmission, but it is still in the FM mode. You have to exit FM mode to "unlock" normal operation. Could it be improved? Maybe, but FM mode is just an extra function, it works decently enough, that would be way too much hustle for way too little gain.

closed 2024-05-19 by egzumer