egzumer/uv-k5-firmware-custom #562: TX/RX backlight without timeout

yarda, 2024-05-08:
It would be great to have TX/RX backlight without timeout or with separate timeout.

E.g. I have 5 seconds timeout which is good for battery saving when interacting with the radio, but during TX/RX the backlight switches off after 5 seconds, which is quite annoying and I e.g. cannot read the s-meter. Workaround is to set higher timeout.

yarda, 2024-05-08:
Or maybe the backlight counter could start after the TX/RX condition is cleared (if the TX+RX backlight is enabled).

Ni5ei, 2024-09-16:
I have side button 2 set to do nothing on short press. I know it's not what you're looking for but at least it lets you wake up the backlight without doing anything else.