egzumer/uv-k5-firmware-custom #579: Fast Copy / Frequency Copy doesn't work for some frequencies

kubedzero, 2024-05-26:
The one-click channel copy feature works fine for UHF frequencies around 440MHz, but not for VHF frequencies around 160MHz. I couldn't find anyone else reporting this issue or mention of this in the wiki so I wanted to raise awareness about this issue. 

In the video, I transmit from a TIDRADIO TD-H3. The Quansheng UV-K5(8) is running the latest version of Egzumer, 0.22. 

Are there limitations to what frequencies the fast copy feature works on, or is this a firmware bug that maybe isn't scanning the correct frequencies?

https://github.com/egzumer/uv-k5-firmware-custom/assets/51148401/01e1ddcc-09f9-4485-9355-c20017daef5e

Happy to provide CHIRP configs, screenshots, or any additional details.

prokrypt, 2024-05-27:
Try:
* setting a lower transmit power
* putting more distance between the radios

ldl805, 2026-04-10:
Can confirm , I just discovered this too. Still applicable