egzumer/uv-k5-firmware-custom #582: physical buttons not working

Berrnard17, 2024-06-02:
tried v22, v21, v17 and 0.1

everything looks normal on screen but none of the physical buttons on the radio do anything.
the radio still reads and uploads with chirp
loaded a cec_0.2 firmware(the waterfall firmware) and buttons now work.  but this fw does not work with chirp 
using a k6 radio.  v22 was working yesterday but i left the radio on overnight.  not sure why that would cause this

Berrnard17, 2024-06-02:
now i have tried yours, uvmod and kitchen fw.  the only one that makes my radio work as intended was the cec/waterfall fw

CobreDev, 2024-07-01:
Also having this issue after flashing v0.22

avrb-studio, 2025-06-07:
Got the same bug on uv-k6, everything looks fine, but buttons not working. Chirp only reads can't write.