egzumer/uv-k5-firmware-custom #589: Step width 8.33 kHz treated differently in regular scanning and spectrum analyzer

catequalsgood, 2024-07-01:
This might be intentional but I noticed that in regular scanning mode a step width of 8.33 kHz always leads to 25 kHz increments after three steps, as expected. But in the spectrum analyzer the step seems to literally be treated as 8.330 kHz. This causes a growing mismatch between scanned frequency and air band channel frequency. 
The attached image shows the result of entering the spectrum analyzer with a scanning rage of 118.000 to 137.000 MHz showing invalid air band frequencies for both the current and highest frequency.

[image]

prokrypt, 2024-07-02:
Definitely not supposed to happen. Someone forgot to `FREQUENCY_RoundToStep()` in a few places.

prokrypt, 2024-07-04:
> The upper limit is now displayed as "136.99999" but I can live with that. Other than that it seems to be working perfectly. Thanks!

Indeed. I figured it wasn't worth using extra bytes to fix such a small quirk. Anyways, on my fork of the firmware the "last frequency" (137MHz) is not scanned anyways.