egzumer/uv-k5-firmware-custom #590: Github Codespace build method - firmware section `.text' will not fit in region `FLASH'

gusosborne, 2024-07-02:
Hello, 

When following instructions for Github Codespace build method, I get the error below. The only change I am attempting to make is to enable NOAA (0 > 1) in makefile. 

----------
/usr/lib/gcc/arm-none-eabi/14.1.0/../../../../arm-none-eabi/bin/ld: firmware section `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/14.1.0/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 644 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:441: firmware] Error 1
----------

Is there possiblly a step omitted from the instructions, or am I doing something wrong?

Thank you,
Gus

prokrypt, 2024-07-02:
compiled firmware too large; disable another feature or two to make it smalle.

gusosborne, 2024-07-02:
OK, that makes sense, but.. Should the instructions not work as written? 
Is there no way to make one singe change without learning and tweaking through all possible options?

Thanks!
Gus

prokrypt, 2024-07-02:
There's a reason NOAA was disabled: to save space to use for another feature.

Anyways, I have discovered NOAA mode is not really needed. I can just store all of its frequencies in memory and tune to them when needed. To prevent the scan function from stopping on those channels, I set a ctcss tone, so when I need to listen to them I just enable monitor mode.

gusosborne, 2024-07-02:
Aha- Thank you, that's a great idea. I'll do that until I can dig into the options..

gusosborne, 2024-07-02:
I do still wonder if the documentation should mention something like: "You may need to disable some features to conserve space, if you add any new options".

closed 2024-07-02 by gusosborne

gusosborne, 2024-07-03:
Just in case anyone else runs across this: I ended up disabling VOX, in order to gain the space to enable NOAA.

ENABLE_NOAA                   ?= 1
ENABLE_VOX                    ?= 0