egzumer/uv-k5-firmware-custom #591: Feature request NOAA and Spectrum Analyzer

PilotBob42, 2024-07-15:
I live in the States and the NOAA feature is still a nice to have here. It, of course, launches off the F+5 combo from the keypad, but the spectrum analyzer feature is also tied to the F+5 button. So, even if you sacrifice something else to make enough room in the build to have both, there is no way to launch both.

How hard would it be to change the launching of the spectrum analyzer to one of the programmable options for side button 1 or 2, or perhaps even long press M? That way the keypad would stay correct as marked for NOAA, but you could still launch the analyzer?

PilotBob42, 2024-07-15:
I dusted off some very very old C coding skills and worked out a way to get what I wanted (NOAA + Spectrum simultaneously). I moved the spectrum analyzer and scan range features from the "5 NOAA" button to the "* Scan" button. This hides the regular scan feature and the CTCSS scanner, but having the VFO settable scan range is better and the "4 FC" function will scan freq and CTCSS so the latter was redundant. Also had to fix up the code a bit around the "Key_5 case in processFKeyFunction of main.c, but I now have factory standard NOAA function PLUS the spectrum analyzer and scan range feature. Unfortunately, to make the 60k limit, I had to dump DTMF calling from the build to make room (never use it anyway).

Did all this a bit quick and dirty since it was just for me, but if anyone is interested in this small but useful change for folks in range of a NOAA station, let me know and I'll clean it up more and perhaps submit it for review.

closed 2024-07-15 by PilotBob42

CodeAsm, 2025-02-04:
Sounds like a way to configure under which buttons or options certain calls to be made would be elegant. I should look in the C myself to look :D (You dont seem to have publicly visible repo's, would be cool to see what youve changed?) have a wonderfull day :D