egzumer/uv-k5-firmware-custom #609: Scan Resume CARRIER issue

Spirit58NL, 2024-09-04:
Hi there!
New here! Just got the QS UV-5K for one week, but with Egzumer it works great! 
One thing: When I set the Scan Resume Mode (Menu22) to CARRIER, it still waits like 5 seconds to carry on scanning, so like TIMEOUT. Do I see that right?

Tnx! Eric.

Cornball5569, 2025-03-14:
Hi @Spirit58NL,

Yes, you see that right (almost). Timing the scan resume time I determined a period of about 4 seconds. Searching the code for the exact time I came up empty handed but stumbled across the answer in this discussion https://github.com/egzumer/uv-k5-firmware-custom/discussions/222

Scan resume time is set to 3.6 seconds:

https://github.com/egzumer/uv-k5-firmware-custom/blob/13b6bb06b22f5c975c2f8295720b3203ba7d479a/misc.c#L63

Hope this sheds some light on the matter.

Cheers.

Spirit58NL, 2025-03-14:
Thanks Cornball5569!

Cheers yourself!

Met vriendelijke groeten,

€ric van den Bosch.