egzumer/uv-k5-firmware-custom #611: Unable to flash with browser on bootloader V5

mlapalme99, 2024-09-09:
Unable to flash with browser on bootloader V5 had to use K5TOOL to Flash

krzysztofkuczek, 2024-09-16:
Probably you have problems with the comm port driver (win11?). If so try to use older drivers http://support.dlink.com.au/Download/download.aspx?product=DGS-3630-52TC (1.12)

mlapalme99, 2024-09-17:
> Probably you have problems with the comm port driver (win11?). If so try to use older drivers http://support.dlink.com.au/Download/download.aspx?product=DGS-3630-52TC (1.12)

It’s definitely a boot loader problem the new boot loader does something funky with the eprom, buddy who made K5TOOL already has a fix and I was was able to finally flash from command line